offspot/kiwix-hotspot #681: Cant Execute kiwix-hotspot on Raspberry Pi

pologoalie8908, 2023-06-06:
When i download the hotspot on my Raspberry PI, and extract it. I Double click on the kiwix-hotspot executable and i get a popup to Execute, Execute in Terminal or Cancel. When i execute, nothing happens.

https://download.kiwix.org/release/kiwix-hotspot/kiwix-hotspot-linux.tar.gz

rgaudin, 2023-06-06:
@pologoalie8908 I think you're not clear on what this file is about. This is an x86_64 executable meant to be used to create a disk image that you can then flash into a µSD card to insert into your RPI.

This _kiwix-hotspot_ (that we are renaming Kiwix Hotspot Imager for this reason) does not run on the Pi itself.

pologoalie8908, 2023-06-06:
So i have a 512GB SD card and i installed Raspbian and some Ham Radio SDR apps on it already. Kiwix is a whole Raspberry Pi OS? Can i install other apps on it after it is configured? Or is it limited to only providing Kiwix stuff?

kelson42, 2023-06-06:
It is limited to what you have chosen to install with the kiwix hotspot imager.

rgaudin, 2023-06-06:
The resulting system is a raspbian as well so should you plug a screen and keyboard you can connect (pi / raspberry) to the terminal and customize as you please.