offspot/kiwix-hotspot #682: Kiwix-server search bar fails to appear on top of screen

Popolechien, 2023-06-08:
From a user feedback: feel free to ask for more info

- Downloaded the pre-made Wikipedia hotspot image;
- Flashed image onto SD card, plugged raspberry pi zero W2
- Search bar fails to appear, be it on Windows or Mobile device (see images)

[image]

[image]

[image]

rgaudin, 2023-06-08:
Downloaded the WP image and I can reproduce.

Kiwix-serve is not launched with the disable taskbar option though:

```
pi@wikipedia:~ $ ps aux |grep kiwix
user       371  0.0  0.0   1940   376 ?        Ss   13:04   0:00 /bin/sh -c ls /data/packages/*.zim && /usr/local/bin/kiwix-serve --threads=8 --port=8002 $(ls /data/packages/*.zim) || /usr/local/bin/kiwix-serve --port=8002 --library /data/packages/library.xml
user       379  0.5  0.2  89076  9940 ?        Sl   13:04   0:01 /usr/local/bin/kiwix-serve --threads=8 --port=8002 /data/packages/phet_en.en.zim /data/packages/raspberrypi.stackexchange.com_en_all.en.zim /data/packages/wikipedia_en_top_nopic.en.zim
pi        1510  0.0  0.0   7352   552 pts/0    S+   13:10   0:00 grep --color=auto kiwix
pi@wikipedia:~ $ /usr/local/bin/kiwix-serve --version
kiwix-tools 3.4.0

libkiwix 12.0.0
+ libzim 8.1.0
+ libxapian 1.4.18
+ libcurl 7.67.0
+ libmicrohttpd 0.9.72
+ libz 1.2.12
+ libicu 58.2.0
+ libpugixml 0.12.0

libzim 8.1.0
+ libzstd 1.5.2
+ liblzma 5.2.6
+ libxapian 1.4.18
+ libicu 58.2.0
pi@wikipedia:~ $ /usr/local/bin/kiwix-serve --help
Usage:
	kiwix-serve [OPTIONS] ZIM_PATH+
	kiwix-serve --library [OPTIONS] LIBRARY_PATH

Purpose:
	Deliver ZIM file(s) articles via HTTP

Mandatory arguments:
	LIBRARY_PATH		XML library file path listing ZIM file to serve. To be used only with the --library argument.
	ZIM_PATH		ZIM file path(s)

Optional arguments:

	-h, --help		Print this help

	-a, --attachToProcess	Wxit if given process id is not running anymore
	-d, --daemon		Detach the HTTP server daemon from the main process
	-i, --address		Listen only on this ip address, all available ones otherwise
	-M, --monitorLibrary	Monitor the XML library file and reload it automatically
	-m, --nolibrarybutton	Don't print the builtin home button in the builtin top bar overlay
	-n, --nosearchbar	Don't print the builtin bar overlay on the top of each served page
	-b, --blockexternal	Prevent users from directly accessing external links
	-p, --port		TCP port on which to listen to HTTP requests (default: 80)
	-r, --urlRootLocation	URL prefix on which the content should be made available (default: /)
	-s, --searchLimit	Maximun number of zim in a fulltext multizim search (default: No limit)
	-t, --threads		Number of threads to run in parallel (default: 4)
	-v, --verbose		Print debug log to STDOUT
	-V, --version		Print software version
	-z, --nodatealiases	Create URL aliases for each content by removing the date
	-c, --customIndex	Add path to custom index.html for welcome page
	-L, --ipConnectionLimit	Max number of (concurrent) connections per IP (default: infinite, recommended: >= 6)

Documentation:
	Source code		https://github.com/kiwix/kiwix-tools
	More info		https://wiki.kiwix.org/wiki/Kiwix-serve
```

What happens is that we did not change our Homepage generator when we updated to kiwix-tools 3.4.0 which introduced URL changes:

`/my-content/` used to (conditionally) render the ZIM and the toolbar. Now it redirects to the toolbar-less `/content/my-content`. The conditional toolbar UI is now at `/viewer#my-content`.

We need to change the URLs in the homepage to point to viewer-based ones.

kelson42, 2023-06-08:
@rgaudin seems super urgent to fix to me.

closed 2023-06-08 by rgaudin

Popolechien, 2023-06-08:
thanks for the fix. Can we / should we relaunch the creation of images?

rgaudin, 2023-06-08:
Relaunched the demo. Will verify on device that it is fixed and relaunch the other auto-images

rgaudin, 2023-06-13:
Fixed ; a new kiwix-hotspot release has been made and is now in use on the Cardshop. The 11G demo has already been updated and tested. The other auto-images are being redone right now.

commit 2ca8cb3b (2023-06-08): Fixed #682: accomodate kiwix-serve 3.4+ URLs

- Chrome URLs are now at `/viewer#{id}`
- Illustration meta are now at `/raw/{id}/meta/Illustration_48x48@1`