offspot/kiwix-hotspot #684: Made image on Card but cant log in on Raspberry Startup

QuantumRodeo, 2023-08-29:
I made two images now from the kiwix card store and when I run the Pi it take me to a login in command prompt. I did not give it a password but it wont let me go to post

rgaudin, 2023-08-29:
Login is default raspbian:
- username: pi
- password: raspberry

closed 2023-08-29 by rgaudin

Popolechien, 2023-08-29:
@QuantumRodeo Just in case I would add that you can only access you content via wifi, plugging your Pi to a screen is not the way to do it (see #683 )

QuantumRodeo, 2023-08-29:
thank you both very much. I am trying to build a off grid kiwix box with touch screen. So i really want to run a screen with it.

QuantumRodeo, 2023-08-29:
any suggestions on how to do this?

Popolechien, 2023-08-29:
I remember seeing some dude building something along these lines. No guarantee it'll work, but the software has much changed on our side so you should be good: https://www.youtube.com/watch?v=R63x2TXm0s8
Other than that `kiwix` + `offgrid` seem to be good keywords to look for.