offspot/kiwix-hotspot #688: Hotspot installer isn't working

LuxStatera, 2023-10-20:
I am trying to use the hotspot installer to build an image for my raspberry pi. When I click the "static content" to access the content catalog, it fails. It says "Catalogs Download Failed" and says I should check my internet connection and/or proxy settings. I tried the installer on macos and linux but both were unable to open the content catalog. Also it seems the windows version is not longer available. How do I solve this?

rgaudin, 2023-10-21:
@LuxStatera thank you for the report. This looks like #677 where I see you already commented.
Can you tried to access that link I mentioned in your web browser? What country are you located in?

rgaudin, 2023-10-21:
Please respond in #677 as this is a duplicate

closed 2023-10-21 by rgaudin