offspot/kiwix-hotspot #690: No Rasberry Pi installer for Windows

rattenbo, 2023-10-26:
Maybe I have missed it, but when I look for the installer for Windows, I only see ones for MAC and Linux.  The only link for windows gets you the Electron version from the Windows Store, which doesn't allow you to create a RASPI SD Card. Any Sugestions?

closed 2023-10-26 by rgaudin

Popolechien, 2023-10-26:
@rgaudin I've removed the Windows installer as my understanding was that it was deprecated / not maintained until the new installer is released?

rattenbo, 2023-10-26:
> That's a website issue ***@***.*** <https://github.com/Popolechien> I
> believe you fixed that today).
> Downloads are available at
> https://download.kiwix.org/release/kiwix-hotspot/

Thanks!

rattenbo, 2023-10-26:
The latest 2 (2.4.12 and 2.4.13) produce the following error.

[image: image.png]

Popolechien, 2023-10-26:
@rattenbo We can't see the images when you reply via email :-/

rgaudin, 2023-10-26:
> @rgaudin I've removed the Windows installer as my understanding was that it was deprecated / not maintained until the new installer is released?

Yes, we should discuss this matter as we'll have no direct replacement for a long while

rattenbo, 2023-10-26:
Popo,

No worries. The text of the message is as follows.

UNHANDLED EXCEPTION IN SCRIPT
X Failed to execute script '_main_' due to unhandled exception
OpenSSL 3.0's legacy provider failed to load. This is a fatal error
by default, but cryptography supports running without legacy
algorithms by setting the environment variable
CRYPTOGRAPHY_OPENSSL_NO_LEGACY. If you did not expect
this error, you have likely made a mistake with your Open SSL
configuration.

Traceback (most recent call last):
  File "_main_.py", line 15, in <module>
  File "runpy.py", line 227, in run_module
  File "runpy.py", line 86, in  _run_code
.
.
.

===================================
Please note that v 2.4.11 DOES work without this error while 2.4.12 and
2.4.13 do generate this error.

I hope that helps.

Thanks,
Richard Attenborough