offspot/kiwix-hotspot #697: Kiwix: Building PI IMG using Linux Binaries (kiwix-hotspot)

kelson42, 2024-02-07:
From kiwix-desktop created by [enema-combatant](https://github.com/enema-combatant): kiwix/kiwix-desktop#1028  

Hi all!  Wonderful product, just trying to build my Raspberry PI image for an SDCARD for the Wifi Hotspot version.

Using this:

https://download.kiwix.org/release/kiwix-hotspot/kiwix-hotspot-linux.tar.gz

Steps to repro:
(1) Execute kiwix-hotspot (runs fine)
(2) Configure options
(3) Hit "Run Installation"

Error is [snipped] as follows:

_

```
[11:10:49 AM] homepage saved to: /tmp/tmpffv0pdt2.html
[11:10:49 AM] [STAGE 1/6: Retrieve Master Image - 0%]
[11:10:49 AM] --> Retrieving base image file
[11:10:49 AM] calculating sum for /run/media/username/Archive-A/kiwix-hotspot.cache/kiwix-hotspot_master_2021-11-24.img.zip...[11:10:50 AM] MATCH.
[11:10:50 AM] Reusing already downloaded base image ZIP file
[11:10:50 AM] [STAGE 1/6: Retrieve Master Image - 8%]
[11:10:50 AM] --> Extracting base image from ZIP file
[11:11:04 AM] --> Failed: [Errno 27] File too large
[11:11:04 AM] Installation failed: [Errno 27] File too large
[11:11:04 AM] [STAGE 1/6: Retrieve Master Image - 17%]
[11:11:04 AM] 
--- Exception Trace ---
Traceback (most recent call last):
  File "zipfile.py", line 1702, in _extract_member
  File "shutil.py", line 208, in copyfileobj
OSError: [Errno 27] File too large

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "run_installation.py", line 187, in run_installation
  File "backend/download.py", line 269, in unzip_file
  File "zipfile.py", line 1630, in extract
  File "zipfile.py", line 1702, in _extract_member
**OSError: [Errno 27] File too large**
```

_

Any ideas?  I have plenty of space on the BUILD directory, and plenty of space on the SDCARD.

Thanks!

kelson42, 2024-02-07:
@enema-combatant Thank you for the bug report but this is the wrong repository. Should be open at https://github.com/offspot/kiwix-hotspot.

rgaudin, 2024-02-07:
@enema-combatant would it be that your BUILD directory is on a FAT32 filesystem or any fs not supporting an 8GB file?

kelson42, 2024-03-05:
Unfortunately no feedback, therefore closing the issue. Please reopen if you come back @enema-combatant

closed 2024-03-05 by kelson42