offspot/kiwix-hotspot #698: Wifi password is set but Wifi connection is showing up as unsecured?

brandonpsmith, 2024-02-22:
I used the paid Kiwix Imager service to build my image and the Raspberry Pi Imager to write to storage. The SSID is correct but shows unsecured? I am unable to connect to the hotspot. Can you help me troubleshoot this please?

rgaudin, 2024-02-22:
This means you chose not to set a passphrase for your network in Imager Service (that's the default). Most platforms consider it unsecure as anyone can connect to it.
It should not prevent you from connecting though.

If you want to solve the passphrase thing just edit your configuration to add a password and reorder. If it was large and you want to save the download time I can post instructions.

You can also provide more details on your connection issue as this should work anyway.

brandonpsmith, 2024-02-22:
I had a password set in this section for my configuration. The image is large. I'm on a mac 14.1.1 can I build on my machine?

I was able to see the SSID. I was able to connect to it (unsecured) and get an IP address. But was unable to connect to ...

http://goto.kiwix.hotspot
http://kiwix.hotspot
http://\<name>.hotspot

[image]

rgaudin, 2024-02-22:
Ok I'll investigate and keep you posted.

brandonpsmith, 2024-02-22:
> If it was large and you want to save the download time I can post instructions.

Can I get these instructions please? I'm going to build an image without a password for now to test.

rgaudin, 2024-02-23:
Here are the instructions to set a password on your WiFi

1. Insert your card into your computer
2. In Finder, locate the `bootfs` volume
3. Find the file named `offspot.yaml` and Open it with a TextEdit (or another **text editor**, not a word processor)
4. Keep the `---` line and right below, paste the following.
5. Replace **(twice)** `abcdefgh` with your wanted SSID
6. Replace `my-secret-password` with your wanted password. Keep in mind there are restrictions, so use something simple. 8 chars min ; You can use imager-service to validate the password (on config save, it will complain if password is invalid).
7. File –> Save to make sure it's saved. Exit TextEdit
8. In Finder, eject the volume. If asked to, select “Eject All”
9. Insert your card in the Raspberry and wait a minute

* [image]
* [image]
* [image]
* [image]

```yaml
ap:
  domain: abcdefgh
  welcome: goto.kiwix
  tld: hotspot
  ssid: abcdefgh
  passphrase: my-secret-password
  as-gateway: true
  dhcp-range: 192.168.2.129,192.168.2.254,255.255.255.0,1h
  network: 192.168.2.0/24
  address: 192.168.2.1
```

brandonpsmith, 2024-02-24:
If you look at my order on Feb 21st you can see it has a Wifi Password set. I can see it in the "Toggle config display" JSON. The property is "wifi_password" and it is not set to null. It has a valid password. This is the image I downloaded via Torrent. The other orders I have not downloaded yet.

When I open the `offspot.yaml` the only line I see is the `---` there is nothing underneath it.

I do not see the captive portal when connected.

[image]
[image]

rgaudin, 2024-02-24:
> If you look at my order on Feb 21st you can see it has a Wifi Password set. I can see it in the "Toggle config display" JSON. The property is "wifi_password" and it is not set to null. It has a valid password. This is the image I downloaded via Torrent and have written to my SD card. The other orders I have not downloaded yet.

Thank you for those details. That's correct indeed.

> When I open the offspot.yaml the only line I see is the --- there is nothing underneath it.

That's what's expected. The rest is to be copy-pasted.

---

Thank you again for the feedback. Those are two separate issues: the SSID passphrase not being passed down the chain and what looks like a firewall issue (on the hotspot) preventing the captive portal from working.

I'll come back to you soon on the first one.

Regarding the second one, can you tell me which Raspberry model you are using and if it is connected to a network via Ethernet?

brandonpsmith, 2024-02-24:
Pi Zero 2 W with a PiSugar 3. It's not connected to any network via ethernet cable.

rgaudin, 2024-02-24:
Oh my, @brandonpsmith I'm really sorry I realize we inadvertently added `Pi Zero 2 W` to the [list of supported devices](https://kiwix.org/en/wifi-hotspot/) (in the FAQ) while we don't.
We definitely intend to but it requires some optimization (the issue is the limited RAM) and we decided this will be done after a few higher priority tasks.

@Popolechien, please initiate a refund.

@brandonpsmith, can we contact you once we're ready to support the PiZero2W? It should be in a couple months I believe. You can subscribe to https://github.com/offspot/base-image/issues/54 if you want.
BTW, the wifi_password thing has been fixed ; thank you again.

@Popolechien, I just updated the FAQ directly.

brandonpsmith, 2024-02-24:
Yes you can contact me no problem. I'm subscribed to https://github.com/offspot/base-image/issues/54 

> BTW, the wifi_password thing has been fixed ; thank you again.

With the Wifi Password issue fixed do I need to issue a new order and rebuild another image? I'll order another Pi to run it on for now.

Is there a Kiwix Slack or Discord where I could send you a DM?

rgaudin, 2024-02-24:
> With the Wifi Password issue fixed do I need to issue a new order and rebuild another image? I'll order another Pi to run it on for now.

Yes. Or use the offspot.yaml instructions I sent earlier.

> Is there a Kiwix Slack or Discord where I could send you a DM?

We're all on [Slack](https://kiwixoffline.slack.com). No Discord AFAIK.

brandonpsmith, 2024-02-24:
The Kiwix Slack is closed. Is there a public email? Do you get messages from the contact form https://kiwix.org/en/contact-us

rgaudin, 2024-02-24:
I believe outside people have a way to join because we frequently get questions there but I don't know how it works. @Popolechien ?

reg at kiwix.org to join me directly.

Popolechien, 2024-02-24:
Yes @brandonpsmith ping us from the contact form and we'll send you an invite link.

kelson42, 2024-04-26:
@rgaudin @Popolechien Can we close this ticket?

closed 2024-04-26 by rgaudin