offspot/kiwix-hotspot #701: Hotspot Companian app - API

Rancho-rachit, 2024-02-28:
- APIs are need for the bridge connection between the 
  - mobile app(<future_build> - Hotspot Companian)
  - kiwix-hotspot
  - Central server

- Requirements -
  - File management
    - get the list of zim files
    - adding zim files
    - deleting zim files
  - metrics
    - Fetch from the Hotspot
    - Upload it to the central server

Rancho-rachit, 2024-02-29:
For the zim file management (tinyfilemanager)

we can use http package from flutter.

Rancho-rachit, 2024-03-05:
- There is no official API for tinyfilemanager - [ISSUE](https://github.com/prasathmani/tinyfilemanager/issues/1104)
- We can use (web_scrapping) for 
  - login
  - listing the content
  - changing the 
    - username
    - password
  
---
- to find, how to
  - upload files
  - use progress bar

rgaudin, 2024-03-05:
@Rancho-rachit please subscribe to https://github.com/offspot/overview/issues/3 ; it looks like we won't need to interact with tinyfilemanager

closed 2024-03-05 by rgaudin